rusefi/alphax-4chan #11: rev A test log

rusefillc, 2022-03-29:
works from +12 as arrived from JLC

rusefillc, 2022-03-29:
```
2022-03-29_17_17_18_586: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2022-03-29_17_17_18_586: EngineState: *** Bluetooth module setup procedure ***
2022-03-29_17_17_18_587: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2022-03-29_17_17_18_587: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2022-03-29_17_17_18_587: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
2022-03-29_17_17_19_471: EngineState: Sleeping...
2022-03-29_17_17_20_436: EngineState: Restarting at 9600
2022-03-29_17_17_20_546: EngineState: Sending AT+VERSION
2022-03-29_17_17_21_521: EngineState: Running at 9600
2022-03-29_17_17_21_522: EngineState: Sending AT+BAUD8
2022-03-29_17_17_21_631: EngineState: received 2 byte(s) 1 [+][O]
2022-03-29_17_17_22_609: EngineState: Restarting at 115200
2022-03-29_17_17_22_609: EngineState: Sending AT+NAMEalphax
2022-03-29_17_17_22_718: EngineState: received 2 byte(s) 1 [+][O]
2022-03-29_17_17_23_701: EngineState: Running at 115200
2022-03-29_17_17_23_702: EngineState: Sending AT+PIN1234
2022-03-29_17_17_24_790: EngineState: Restarting at 38400
2022-03-29_17_17_24_791: EngineState: SUCCESS! All commands (4 of 4) passed to the Bluetooth module!

```

rusefillc, 2022-03-29:
with R40 removed and 115 added have +5v made from +12v

rusefillc, 2022-03-30:
https://github.com/andreika-git/hellen-one/issues/148 reported

rusefillc, 2022-03-30:
for CAN sleep/wake up see https://github.com/rusefi/alphax-4chan/issues/13

rusefillc, 2022-03-31:
baro works - reads 101kPa

ETB works!!!

(I still hate soldering ETB chips - first it worked for a second and scared shit out of me. works much better once more properly soldered)

andreika-git, 2022-03-31:
> for CAN sleep/wake up see #13

A proper (hopefully?) CAN wakeup with an additional inverted buffer is on 2ch-revC, not on 4ch-revA.

rusefillc, 2022-04-26:
https://rusefi.com/online/index.php?vehicleName=La+Mitsubicha&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

https://youtu.be/egtDQsMDQqs

closed 2022-05-15 by rusefillc