rusefi/alphax-4chan #14: Vertical USB

rusefillc, 2022-03-31:
Considering case layout vertical would work better

[image]

rusefillc, 2022-03-31:
@ElDominio what do you think?

Vertical usb is a bit more fragile, probably no space for usb-b?

ElDominio, 2022-03-31:
While I understand, I typically drill a hole for people demanding access, it can be done but I don't know if it's worth it.

USB-B is the printer one right?

rusefillc, 2022-03-31:
yep, USB-B is the printer one

vertical connector is about 14x14mm which we do not have

ElDominio, 2022-03-31:
Yeah, if we can switch to type B vertical, it could be done, but if we keep miniusb I'd say keep horizontal.

andreika-git, 2022-03-31:
Unfortunately, it's highly unlikely that we can find more space to fit the USB-B-Vertical connector without total rearrangement of the whole board... Unless we remove the left mounting hole (near the bluetooth module)...