rusefi/alphax-4chan #15: Silkscreen IGND reads as "I-GND" not as "IGN-D"

rusefillc, 2022-04-01:
Also schematics says IGN4

let's change silk to follow schematics - from IGNA-D to IGN1-4

andreika-git, 2022-04-01:
It was inherited from the original board:
[image]
Ok, let's change it to our numerical notation!

rusefillc, 2022-04-01:
@ElDominio please approve IGN4

ElDominio, 2022-04-01:
I approve this, it's better.

andreika-git, 2022-04-01:
Do we need the same for INJA..INJD?

rusefillc, 2022-04-01:
@ElDominio i guess yes on 4chan?

rusefillc, 2022-06-10:
schematics says 1-4 for both

[image]

let's replace A-D with 1-4 on all silk

andreika-git, 2022-06-10:
[image]
[image]

Done
https://github.com/rusefi/alphax-4chan/commit/a3c0af0193ebfe586d35963ddcb781002348c70c

closed 2022-06-10 by rusefillc