rusefi/alphax-4chan #19: Q3 hardware mod for rev A and rev B

rusefillc, 2022-05-02:
See https://github.com/rusefi/alphax-2chan/issues/53

andreika-git, 2022-05-03:
There is also a suggestion to swap p-channel mosfets for vishay SI2347DS-T1-BE3 is similar but has 20V max V_gs part:
* Q3, Q4, Q5, Q16, Q17, **Q31, Q32**.

rusefillc, 2022-05-04:
@andreika-git do we have JLCPBC suggestions maybe it's not too late to replace part in the current order

andreika-git, 2022-05-04:
SI2347DS is perfectly available on JLC: C710065 or C145002

rusefillc, 2022-05-10:
I've ordered all qty 14 of SI2347DS-T1-BE3 available on Mouser

need more to fix the 10 units of rev A I am receiving

rusefillc, 2022-05-11:
@andreika-git says 726-BSS308PEH6327

rusefillc, 2022-05-19:
@andreika-git just to confirm that Q18 is NOT affected?

andreika-git, 2022-05-19:
> @andreika-git just to confirm that Q18 is NOT affected?

Yes, it has 3.3v which is < 12v

closed 2022-07-26 by rusefillc