rusefi/alphax-4chan #20: rev B test log

rusefillc, 2022-05-15:
does not DFU as arrived from JLC

This is different from https://github.com/rusefi/alphax-2chan/issues/47 since no USB power surge and a number of LEDs are ON, just no DFU

rusefillc, 2022-05-15:
With USB power, 3v regulator gets 3.6v input instead of expected 4.8v input

with 3.6v input 3v generator outputs about 0.1v

rusefillc, 2022-05-17:
keweisi usb detector says 1.6A

hottest area SEEMS to be between MCU module and VR module but hard to say with more precision :( i think back side gets cooler when I touch it, i take all the heat?

rusefillc, 2022-05-17:
3v regulator U101 seemed like the hottest part

I've removed U101 and consumption went down to 0.17A

rusefillc, 2022-05-17:
with U101 removed 3v to ground beeps, seems to be about 1 ohm

rusefillc, 2022-05-17:
found issue it was similar to Harley defect

root cause lack of automated validation https://github.com/andreika-git/hellen-one/issues/152

rusefillc, 2022-05-18:
it was a gerber issues with a via under  C461

[image]

Q: how happy is this ECU going to be without C461?

[image]

rusefillc, 2022-05-22:
gerber issue around Ign fixed by https://github.com/rusefi/alphax-4chan/commit/38b9e8fdefc0ad3ff9537dbe6887440f12287586

rusefillc, 2022-07-02:
three cuts total

[image]

[image]

[image]

rusefillc, 2022-07-11:
[image]

[image]

ElDominio, 2022-07-11:
I need to do this to revB 4chan I suppose?

rusefillc, 2022-07-11:
@ElDominio short answer is NO

ElDominio, 2022-12-03:
Close?

closed 2022-12-03 by rusefillc