rusefi/alphax-4chan #24: More vias under ETB chip

rusefillc, 2022-05-24:
@mck1117 suggests more vias under ETB chip

can we actually have huge holes to facilitate soldering with limited tools?

4chan at the moment
[image]

probably Proteus
[image]

rusefillc, 2022-06-09:
@ElDominio reports issues during calibration, we need an incremental improvement.

```
overheat
a few cars have had issues BUT, not driving
couldn't the underside have a bit of copper to solder it onto for heat dissipation?
```

My question: can we make LARGE vias like 1.5mm to give a chance to solver through back side to make soldering simpler?

ElDominio, 2022-06-09:
About soldering, I have a guy that solders them PERFECTLY now. This improved the situation but did not fix it.

mck1117, 2022-06-09:
Proteus has 3 ground planes, and tons of vias under the chips, and has no overheating problem.

rusefillc, 2022-06-10:
@mck1117 @Qwerty-OFF please review

rusefillc, 2022-06-11:
@andreika-git says that connected polygon could make it too hard to solder in terms of heat requirmen

closed 2022-06-11 by rusefillc