rusefi/alphax-4chan #25: Main Connector Fitment issues - see picture

ElDominio, 2022-06-08:
[image]

The connectors are ever so slightly too close together, making the connectors bend upwards in the middle. They need to be space a tiny fraction of an mm

rusefillc, 2022-06-09:
Million dollar question: how much do we move let's say grey connector?

I have both "legacy" and "neo" 4chan in my hands and maaaaaaaaybe "legacy" is a bit easier to install in the case

@ElDominio I have no idea how to measure the amount we need to move the grey connector (randomly picking which one to move)

0.3mm?

@andreika-git how do gerbers compare between "legacy" and "neo"? is there _any_ difference to justify minor difference which I _maybe_ feel?

ElDominio, 2022-06-09:
I'm thinking maybe split them both, move each like .15mm or so. It's very tiny, but it doesn't allow the top plate to be fully screwed into place, which is a bit problematic

andreika-git, 2022-06-09:
> @andreika-git how do gerbers compare between "legacy" and "neo"? is there _any_ difference to justify minor difference which I _maybe_ feel?

The original gerbers (1200 dpi):

[image]

The 4ch-revb gerbers (1200 dpi):

[image]

The difference (1200 dpi):

[image]

andreika-git, 2022-06-09:
Let's check the bottom-left hole:

1) The original gerbers:
  Origin corner: (91.3, -152.2), Hole location: (96.9, -142.65)
  **Coordinates: 5.6 9.55**
  **Diameter: 1.8** 

2) 4ch-revB:
  Origin corner: (0, 114.7), Hole location: (9.64946, 109.101)
  **Coordinates: 5.599 9.64946**
  **Diameter: 1.50114** 

The max difference between the coordinates is 9.64946-9.55 = **0.1 mm**
Visually I see the max. difference ~5 pixels, which also means ~0.1 mm @ 1200 dpi.

I think the main difference is not coordinates but hole size: let's make 1.8 mm instead of 1.5 mm

P.S. Are all connectors the same?

rusefillc, 2022-06-10:
both connectors overall outer size where it matters: 91.8mm
inner width of relevant case region: 93.1mm

let's move both connectors apart from each over, let's move each by 0.35mm for overall width change of 0.7mm

let's NOT change hole diameter there is no reason for that

andreika-git, 2022-06-10:
done.
https://github.com/rusefi/alphax-4chan/commit/a3c0af0193ebfe586d35963ddcb781002348c70c

We still have some space on the edge:
[image]