rusefi/alphax-4chan #28: rev C BOM state

rusefillc, 2022-06-10:
1. let's remove R40 since we always use TLS115

[image]

2. new alternative for F101
[image]

3. no action needed, just FYI - I am out of parts in library, trying to order more. I have ebay parts

[image]

4. let's remove from BOM to reduce distraction or find some way to mark so that it's less visible

[image]

rusefillc, 2022-06-10:
5. C87403 is in stock but it's 20% of total board price?!

andreika-git, 2022-06-10:
A possible replacement for C87403 is C213161.

mck1117, 2022-06-10:
looks like C213161 is a bit slower, but still probably OK

rusefillc, 2022-06-11:
I now own some C213161 in parts library

rusefillc, 2022-06-11:
DaveB says:

[C92485](https://lcsc.com/product-detail/Analog-Switches-Multiplexers_Texas-Instruments-TS5A3159DBVR_C92485.html) 

C10426 ? C255474 ?

[12:08](https://rusefi.slack.com/archives/CKG9KQHS7/p1654920503226719)
C255474

closed 2022-06-14 by rusefillc