rusefi/alphax-4chan #29: rev C - low stock and not cheap SN74LVC2G17DCKR C50714

rusefillc, 2022-06-11:
any cheaper C50714 alternatives?

[image]

rusefillc, 2022-06-11:
@andreika-git do we have options in same or smaller packages?

andreika-git, 2022-06-11:
The same package, the cheapest but low stock: C94189
The same package, rather expensive but more stock: C80500

rusefillc, 2022-06-11:
i now own x65 of C94189

rusefillc, 2022-06-11:
Dave mentions [C458779](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C458779)

this nexperia partis cheap ($0.16) on a reel.  depending on how many you use, may be worth global sourcing  supply sucks right now, but sourceengine has a ton. mouser has 80ish

extreme part shortage of all parts in this package.  I would be looking at alternative packages or possibly a quad or single vs. dual buffer

rusefillc, 2022-06-11:
New separate issue https://github.com/rusefi/alphax-4chan/issues/32

mck1117, 2022-06-14:
C80500 WILL NOT work - not rated for use of the high side diodes

mck1117, 2022-06-14:
[C458779](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C458779) will not work since it's 3.3v only, and has no high side diodes either.

mck1117, 2022-06-14:
Actually - the original part C50714 doesn't work either since it has no high side diodes!

mck1117, 2022-06-14:
we must use C94189

rusefillc, 2022-06-14:
qty 500 of 74HC2G17GW,125 https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C94189 has arrived to NJ

closed 2022-06-14 by rusefillc