rusefi/alphax-4chan #32: Test +12v Hall input on rev A or B

rusefillc, 2022-06-11:
Suddently there is a concurn that maybe 4chan rev A and B does not support +12v Hall Input. We need to either fix boards or attempt potentially destructive test

rusefillc, 2022-06-11:
So I configure Hall mode on 4chan rev A or B board, feed +12v from 1A power supply. 

Worst case scenario how much of the board would we fry? is there a chance to fry stm32?

@ElDominio just FYI

rusefillc, 2022-06-13:
@andreika-git what is the bad case scenario in terms of specific board damage?

andreika-git, 2022-06-13:
> @andreika-git what is the bad case scenario in terms of specific board damage?

The worst case scenario is U2/U5 failing and passing +12V directly to the digital inputs of the mcu, and it would fail too (5v-tolerant pins of STM32 don't have overvoltage protection diodes).