rusefi/alphax-4chan #34: where are TPS/PPS pull-downs?

rusefillc, 2022-06-12:
[4:44 PM, 6/12/2022] Norman AplhaX rusefi PR: i have one question, i'm trying to find my answer in the schematic but i can't open the part about analog inputs

do the analog inputs have tiny (100kohm) pulldowns/pullups to detect open circuit?
[4:44 PM, 6/12/2022] And: yes and no
[4:44 PM, 6/12/2022] And: no and yes
[4:44 PM, 6/12/2022] Norman AplhaX rusefi PR: or should those be added externally if such a thing is desired (because i assume it would throw off scaling of analog sensors)
[4:44 PM, 6/12/2022] And: some do have 500K for sure
[4:44 PM, 6/12/2022] And: some have 10K
[4:45 PM, 6/12/2022] And: i guess what exact channels do you care about?
[4:45 PM, 6/12/2022] Norman AplhaX rusefi PR: specifically PPS1/2 and TPS1/2
[4:45 PM, 6/12/2022] And: i bet those have weak pull-downs

andreika-git, 2022-06-13:
TPS1, PPS1: 680k pull-downs
TPS2, PPS2: 10k pull-downs

mck1117, 2022-06-13:
10k is too strong btw, they should be more like the 680k

andreika-git, 2022-06-13:
[image]

rusefillc, 2022-06-13:
@ElDominio FYI