rusefi/alphax-4chan #35: C7/F7 shown as ADC

ElDominio, 2022-06-18:
[image]

I think that's wrong, is it not?

rusefillc, 2022-06-18:
it sounds like technically crank and cam could be used as analog @andreika-git please confirm

```
  - pin: C7
    # H144_IN_CAM
    id: [A6, A6, EFI_ADC_6]
    class: [event_inputs, switch_inputs, analog_inputs]
    function: Camshaft HALL Sensor or Generic Digital Input
    type: hall
    ts_name: C7 - CAM Hall/Digital
    color: brown
```

@ElDominio please clarify if you would rather hide to reduce potential confusion

ElDominio, 2022-06-18:
If they can, that would be cool, but aha, if they can't they shouldn't be listed today. I had a guy ask me if he could use it as such today lol

andreika-git, 2022-06-18:
Yes, C7 can be used as a generic analog input.

F7 can be also used (in theory), but it's special: 
1) it can measure 12V voltage (and more) and thus it has a voltage divider, so we need to modify the firmware to support it as a generic input (add a divider coefficient).
2) VIGN is currently required by the firmware for main relay and power control.
So currently it's not recommended to use F7 as an alternative analog input, although it's possible in the hardware (with a custom firmware).

mck1117, 2022-06-18:
> Yes, C7 can be used as a generic analog input.
>
> F7 can be also used (in theory), but it's special:
>
> 1. it can measure 12V voltage (and more) and thus it has a voltage divider, so we need to modify [...]

It's also on adc3 so we can't actually use it without significant architectural changes. Aka, you can't use it.

andreika-git, 2022-06-18:
> It's also on adc3 so we can't actually use it without significant architectural changes. Aka, you can't use it.

pin C7 = IN_CAM_HALL = IN_CAM = AIN19 = A19 = PA6 = ADC12_IN6
Aka, you can use it.

rusefillc, 2022-07-26:
@ElDominio shall we close this one or still anything actionable?

ElDominio, 2022-07-26:
Still not clear if can use or not as ADC

mck1117, 2022-07-26:
As far as I can tell from the schematic, you can use it as analog.

rusefillc, 2022-07-26:
@ElDominio ^^^

closed 2022-07-26 by rusefillc