rusefi/alphax-4chan #36: Main relay control not working properly

ElDominio, 2022-06-22:
I can't get a log but I can get a video. Main relay control works once, then after turning the key back on, the ECU seems to hang, I cannot connect to it through USB until I unplug it from the car and plug it back in

rusefillc, 2022-06-22:
any board or specific board?

are you personally there?

ElDominio, 2022-06-22:
Yep, it's the Lancer I am testing. RevA 4Chan

rusefillc, 2022-06-22:
Thank you for answering the second question!

Do you have more 4chan units you can try on same vehicle? i.e. is this issue specific to one board or any board is affected?

ElDominio, 2022-06-22:
I do, I'll do that on Thursday too

ElDominio, 2022-06-23:
About this:

Apparently, when battV is below a certain point, main relay control fails. Charged the battery up and now it's working as expected

ElDominio, 2022-06-23:
For some reason, when battery voltage was about 10.8v, the ECU would just hang. Now that the battery is charged it works as expected

rusefillc, 2022-06-28:
@ElDominio close this one?

ElDominio, 2022-06-28:
Will run some more tests tomorrow and comment back on this

ElDominio, 2022-07-06:
Close this since it was discovered this is a Lua issue, not a 4chan issue.

rusefillc, 2022-07-06:
See https://github.com/rusefi/rusefi/issues/4322

closed 2022-07-06 by rusefillc