rusefi/alphax-4chan #38: Tach voltage not +12v

rusefillc, 2022-06-28:
with D1 removed voltage is 9.9v

with R46 value lowered to 470 tach voltage 11.5v

andreika-git, 2022-07-23:
https://github.com/rusefi/alphax-4chan/commit/4de32993da84c7dde138b381038f24e7c7060564

[image]

ElDominio, 2023-02-02:
RevE still not ok

In 12v mode, still spitting out around 4v[image]

rusefillc, 2023-02-02:
Первый шаг отпаять D1 или D9 (светодиод TACH)

[image]

Второй посмотреть осциллографом что происходит на левой ноге R47

rusefillc, 2023-02-02:
Baseline test in NJ

1R load
actual result:
tach LED does not blink
0v<>1v square wave with "5v" tach mode
0v<>2v square wave with "12v" tach mode

rusefillc, 2023-02-02:
removed D9
nothing has changed - same 0/1 or 1/2v levels

left side of R47 constant 12v

rusefillc, 2023-02-02:
1K load was a BAD TEST

without any load, with D9 removed I have happy 0v/+12v on tach output!

now going to return D9 since did not test without load :(

rusefillc, 2023-02-02:
@ElDominio it seems to work fine on bench

@Qwerty-OFF says it could be about your specific resistance on specific tach.

@ElDominio can you please try with tach disconnected - would it be 9v or 12v without tach?
@ElDominio can you measure tach resistance to GND?

ElDominio, 2023-02-04:
Hi, in the local tests I have done:

1) With nothing connected, it behaves as expected
2) With an older 12V tach, voltage drops as seen on the scope, tach does not move
3) Works fine with logic based tachs (5v), since they seem to only use the square wave to drive a motor
4) In 12V cases, we usually use the 5v output to drive a N2222 transistor pulled up to 12v, which then is used to drive the tacho. This works fine

rusefillc, 2023-03-16:
@ElDominio are you looking for any changes on rev G like making the N2222 official right on the board? not sure if things would stay configurable

@ElDominio the safest option would be to provision for maybe even through hole N2222 right on 4chan but have it not populated by default, would you like that?

ElDominio, 2023-03-16:
Having space for a 2222 THT, sounds at least like a _decent_ fix to make it _somehwat_ configurable, and I'd like it in all boards, I had totally forgotten about this

Qwerty-OFF, 2023-03-18:
Now we have a transistor like N2222 pulled up to 5V. It can also be pulled up to 12V by additional control signal from MCU. What needs to be added?
[image]

ElDominio, 2023-03-19:
So when I have issues with 12v tach, I usually use the Tach output on the ECU to drive a 2222 pulled up to 12V on the output, and then this produces a clean 12v square wave

ElDominio, 2023-03-19:
Pullup is 1k

Qwerty-OFF, 2023-04-02:
Configurable pullup:
[image]
Low-side MOSFET:
[image]

rusefillc, 2023-04-02:
Reopening to game more notes

ElDominio, 2023-07-23:
RevG can change voltages just fine

closed 2023-07-23 by rusefillc