rusefi/alphax-4chan #39: randomly does not turn off

rusefillc, 2022-06-29:
https://youtu.be/YtoW3d-4EBI

rusefillc, 2022-06-29:
please connect laptop please tell battery voltage gauge with ign key off

if ign key is off I can offet a Lua hack:

a lua script which
a) waits for battery to go above 11
b) if battery is ever below 5 after it was above 11, put rusEFI to sleep. that would turn of main relay and address back feed

that's probably a hack?

ElDominio, 2022-06-29:
Will update on Thursday then

rusefillc, 2022-06-29:
something along these lines

```
shallSleep = Timer.new()

-- we want to turn on with hardware switch while ignition key is off
hadIgnitionEvent = false

function onTick()

	if vbatt > 11 then
		-- looks like we have ignition key do not sleep!
		shallSleep : reset()
		hadIgnitionEvent = true
	end

	if hadIgnitionEvent and shallSleep : getElapsedSeconds() > 3 then
		-- looks like ignition key was removed
		mcu_standby()
	end

end

```

mck1117, 2022-06-29:
@rusefillc wait, isn't this just the main relay control logic? It turns the main relay off if the ignition voltage drops.

rusefillc, 2022-06-29:
could be :(

TL,DR: i have same/similar drama on proteus. on Proteus I have CAN-based ignition key that helps

mck1117, 2022-06-29:
Where is the power coming from? What pin still has power that we liberate by turning off the main relay?

Also, log?

ElDominio, 2022-06-29:
Log can't show anything imo, if the key is turned off with the laptop, you can see voltage change, but the ecu behaves as expected and shuts off when you disconnect USB.

ElDominio, 2022-06-29:
If the ECU gets stuck on and keyed off, you connect USB and the ECU is just hung, USB Device Descriptor Request Failed shenanigans

mck1117, 2022-06-29:
> Log can't show anything imo, if the key is turned off with the laptop, you can see voltage change, but the ecu behaves as expected and shuts off when you disconnect USB.

If you can see the ignition voltage change and the ECU doesn't turn off the main relay, this is a bug.

mck1117, 2022-06-29:
> b) if battery is ever below 5 after it was above 11, put rusEFI to sleep. that would turn of main relay and address back feed

The main relay control code is supposed to be doing this already

ElDominio, 2022-06-30:
I'm with the car

Voltage detects perfectly but problem continues; again i Can't log it, since with the laptop it can't be replicated

rusefillc, 2022-06-30:
is connector easy to depin? do you know how to depin connector

can you disconect as many low side driver wires as possible? idle? vvt? maybe even relays etc?

i.e. what's the minimal set of wires to reproduce the problem?

and you say it's sporadic not 100% reproducible? :(

rusefillc, 2022-06-30:
crazy ideas:

lua script to turn off if no RPM for 60 seconds?

ElDominio, 2022-07-01:
https://drive.google.com/drive/folders/1_tZcS8Qz7JpopSBae-BIwZ7rOK1sBMPq?usp=sharing

Videos of the issues. @mck1117 helped me trace this back to a Lua issue causing a boot loop. 
My Lua code is here
https://gist.github.com/ElDominio/8385f846d1e0bcc8d2e2569ff62d45e9

rusefillc, 2022-07-01:
Current status unclear.

Is this issue resolved? Is this issue not resolved but solegow narrowed down to specific lua script?

mck1117, 2022-07-02:
Not resolved but it's a little more data. Alpha and I are still looking in to it.

ElDominio, 2022-07-02:
Yeah, right now @mck1117 is waiting on me till I can connect to the ECU with my STlink

I have an STlink V2 btw, not V3, obviously ebay clone. This works right? Where does it connect to?

rusefillc, 2022-07-03:
Г. Б

rusefillc, 2022-07-03:
you have three options https://github.com/rusefi/rusefi/wiki/Tag-Connect-TC2030 describes all three

most proper - order and solder FTSH-105-01-F-DV-007-K connector and v3 which is cheap

proper with zero soldering: v3 + proper tc2030 cable for v3

cheapest: v2 + ghetto tc2030 based on eBay TC2030-MCP-NL with phone jack cut off

rusefillc, 2023-03-14:
@ElDominio how can I reproduce at home? Shall we close or else?

ElDominio, 2023-03-14:
Close for now, Haven't really run into it

closed 2023-03-14 by ElDominio