rusefi/alphax-4chan #43: Can we try TSON-10 TLS115

rusefillc, 2022-07-22:
We plan a smallish batch do we have confidence to go TLS115 TSON-10?

As a module https://github.com/andreika-git/hellen-one/issues/167 or manually?

rusefillc, 2022-07-25:
TLS115D0LDXUMA1
MFR.Part #
TLS115D0LDXUMA1
Manufacturer
INFINEON TECHNOLOGIES
JLCPCB Part #
C9900011264
Description
Linear Voltage Regulator IC Positive Adjustable 1 Output 150mA PG-TSON-10
Datasheet
Min. assembly qty
15pcs
Attrition
4pcs

rusefillc, 2022-07-25:
Let me discuss with @ElDominio

rusefillc, 2022-07-29:
@andreika-git please move forward

andreika-git, 2022-07-29:
[image]

[image]
[image]

"TLS115","U1","hellen-one-common:PG-TSON-10-2","C9900011264"

closed 2022-07-29 by rusefillc