rusefi/alphax-4chan #49: JLC assembled ideas for Schmitt-trigger

rusefillc, 2022-07-25:
We have four channels of Schmitt-trigger 

[image]

now that we have double sided assembly do we have _any_ fully JLC options?

rusefillc, 2022-07-25:
Still related to https://github.com/rusefi/proteus/issues/1 :)

rusefillc, 2022-07-26:
@Qwerty-OFF пожалуйста сначала предложи вариант, но не меняй файлы - там потенциально @andreika-git будет другие изменения делать, нужна будет координация

Qwerty-OFF, 2022-07-26:
74HC14 шестиканальный инвертирующий (видимо придется менять софт, был неинвертирующий)
74HC1G14 одноканальный инвертирующий
Либо ставить двухканальный аналогичный тому что уже есть, но без защиты, защиту поставить отдельно, снизу место есть.

rusefillc, 2022-07-26:
@Qwerty-OFF лучше всего вариант защиту поставить отдельно, снизу место есть.

@andreika-git как скоординировать изменения?

rusefillc, 2022-07-29:
@andreika-git can you add external protection on the back or do prefer that @Qwerty-OFF implements it?

andreika-git, 2022-07-29:
I'll do. Perhaps, I'll use tree BAT54S...

andreika-git, 2022-07-29:
[image]
[image]

# trigger schmidt buffer
"U2,U5","74LVC2G17GW","SOT-363","C80500"

closed 2022-07-29 by rusefillc