rusefi/alphax-4chan #50: VND7NV04TR-E 60ohm/6A instead of VND14NV04 35ohm/12A

rusefillc, 2022-07-26:
VND7NV04TR-E
C155605

to replace hand assembled VND14NV04

rusefillc, 2022-07-26:
@andreika-git please confirm pinout/shape match

@andreika-git please confirm that we are OK with extra resistance

ElDominio, 2022-07-26:
I thought we were waiting to see if JLC got VND14s back in stock?

The eBay sourced ones seem to work well too, I'd rather have the peace of mind of the 14s if possible

rusefillc, 2022-07-26:
a) there is no way that the smaller pins hold even 6A, i bet same applies to larger pins
b) JLC UI gives me an option to preirder VND14 C155647 but those are $4.5 instead of $0.60 ebay, i.e. $40 extra per board

ElDominio, 2022-07-26:
I can have them from eBay and solder here, VND14 is not too difficult to solder
most VND outputs are on the edge pins which hold about 10A

The most important thing is being able to hold 4 injectors per channel cause trust me people will do that

closed 2022-07-26 by rusefillc