rusefi/alphax-4chan #55: two-solenoid IAC dos not work

rusefillc, 2022-08-05:
https://rusefi.com/online/view.php?msq=1138&dialog=idlehw

think there is a problem with two wire idle on 4chan

[andrey](https://app.slack.com/team/U4PVDN5L7)
  [9:00 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661238635819)
are you trying two idle IAC for first time?

[AlphaX](https://app.slack.com/team/U02PS3897M3)
  [9:00 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661259578909)
yep
[9:01](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661279983469)
i think there is some flyback or something, if they are both connected, the idle valve doesn't move
[9:01](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661285007319)
and both bulbs are equally lit
[9:01](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661293782369)
if i disconnect one, it works in the direction that is still connected

[andrey](https://app.slack.com/team/U4PVDN5L7)
  [9:01 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661316103349)
"they are both connected"
[9:02](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661339052339)
any chance you are talking about a three wire IAC?

[AlphaX](https://app.slack.com/team/U02PS3897M3)
  [9:03 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661384058059)
yes that
[9:03](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661388709489)
two wire iac is fine
[9:03](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661396550159)
i forgot three = 2 control wires

[Simon - FutureProof Performance](https://app.slack.com/team/UF9NQSX8T)
:palm_tree:  [9:05 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661536647589)
Double solenoid IAC

[AlphaX](https://app.slack.com/team/U02PS3897M3)
  [9:07 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661634289279)
yes
[9:07](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661651307179)
i'm going to install diodes on it later to see if it started working correctly

[andrey](https://app.slack.com/team/U4PVDN5L7)
  [9:08 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661693689269)
that's all very weird
[9:08](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661704189649)
"install diodes" huh
[9:09](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661751131579)
with a lot of respect your tune contains SO much of data that not providing it is a crime

[Matthew Kennedy](https://app.slack.com/team/U80K9BYVD)
  [9:09 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661765451389)
There are two camps on how to control a three wire
[9:09](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661778156979)
Either you energize both, or you only ever do one at a time

[andrey](https://app.slack.com/team/U4PVDN5L7)
  [9:10 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661808609069)
image.png 
image.png

:+1:
1

[AlphaX](https://app.slack.com/team/U02PS3897M3)
  [9:10 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661815094929)
[@andrey](https://rusefi.slack.com/team/U4PVDN5L7)
 it's on rusefi online

[andrey](https://app.slack.com/team/U4PVDN5L7)
  [9:10 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661816924579)
i do not comprehend "install diodes"
[9:10](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661828472849)
:disappointed:

[AlphaX](https://app.slack.com/team/U02PS3897M3)
  [9:10 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661844551869)
image.png 
image.png

[andrey](https://app.slack.com/team/U4PVDN5L7)
  [9:11 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661888713829)
https://rusefi.com/online/view.php?msq=1138&dialog=idlehw

rusefi.comrusefi.com
[rusEFI online](https://rusefi.com/online/view.php?msq=1138&dialog=idlehw)
rusEFI tune file sharing site
New

[Matthew Kennedy](https://app.slack.com/team/U80K9BYVD)
  [9:11 PM](https://rusefi.slack.com/archives/C4Y7HA6NM/p1659661907662279)
I see diodes

rusefillc, 2022-08-05:
@ElDominio i need the part number so that I could get this stuff here, I can solder some wires to connector hopefully

ElDominio, 2022-08-22:
On another 2 wire IAC, it seemed to work just fine (Toyota) , this might be case-by-case

ElDominio, 2022-12-03:
Close as i have used two wires successfully now