rusefi/alphax-4chan #63: What component to remove to ECU does not stay powered with a constant 12V source on an output

ElDominio, 2022-09-05:
On the PNP unit I made for Lancer X, A/C Clutch control is connected to F1 (Idle2). On the revA 4chan I was instructed to remove a component to make the ECU shut down properly, but I don't remember which one it is. Now that I am swapping to RevC I need to remove it again for the same effect.

rusefillc, 2022-09-05:
@mck1117 were you involved with this? Do you recall what was this about?

Would be nice to get to the more bottom bottom of this :)

ElDominio, 2022-09-05:
[image]

rusefillc, 2022-09-05:
@ElDominio is this A/C clutch PWM or on/off?

Anyway this or similar has happened enough of times that it needs to be understood more completely. Why does it happen for some outputs on some vehicles but no on others?

I am sure I had something like this on NB2 miata VVT, I am sure that I have back-feed on my Passat B6

@andreika-git @Qwerty-OFF @mck1117

ElDominio, 2022-09-05:
On/Off 
Lancer 2009 N/A

The clutch relay has 12v at all times, if D51 is not removed, the ECU gets 12V from this output and turns on. Removing D51 removes this bad behavior, but probably gives problems with PWM. Since this output does not need to be PWM'd, this is fine.

rusefillc, 2022-09-05:
We are lucky here this time but there is the wider pattern.

Is this about A/C being connected to battery directly not sitting behind main relay?

rusefillc, 2022-09-05:
https://github.com/rusefi/rusefi_documentation/blob/master/OEM-Docs/Mitsubishi/2009-Lancer-Enginers-2L-turbo-2L-NA-2.4L-NA.pdf

ElDominio, 2022-09-05:
Stupid Github phone app

Yes, it is because what you said

ElDominio, 2022-12-03:
Can be closed

closed 2022-12-03 by rusefillc

ElDominio, 2023-11-22:
@rusefillc @Qwerty-OFF what is the equivalent of d51 in revH?

Qwerty-OFF, 2023-11-22:
D66 for IDLE2 / F1 pin

rusefillc, 2024-02-27:
https://github.com/andreika-git/hellen-one/issues/368