rusefi/alphax-4chan #66: Mechanical improvements for MAP sensor footprint

rusefillc, 2022-10-05:
We have some boards with MAP pulled off

only three of eight pads are actually used for anything

what tricks like vias and more metal can we try to increases our mechanical chances?

rusefillc, 2022-10-05:
@ElDominio any chance to see photos of damanges PCBs?

rusefillc, 2022-10-05:
@andreika-git please check datasheet if we can GND or else currently disconnected pads

ElDominio, 2022-10-05:
Will upload later today i think they're in my bag

ElDominio, 2022-10-07:
[image]

rusefillc, 2022-11-29:
Hot glue did not help these do not even survive shipping :(

[image]

@timbelom part of the problem is poor soldering?

[image]

mck1117, 2022-11-29:
is that broken solder joints, or pads torn off the board?

dron0gus, 2022-12-01:
Do not put high profile SMD components close to board edges.

rusefillc, 2022-12-01:
> Do not put high profile SMD components close to board edges.

plz remind why :)

mck1117, 2022-12-01:
> plz remind why :)

I seem to remember somebody recently had issues with tall components near the edge recently 🤔

[image]

ElDominio, 2022-12-01:
I assume that having it near the edge means it won't be inside the profile created by the connector and the edges of the board

rusefillc, 2023-01-22:
@Nadezhda-n please match latest 2chan maybe with larger center hole

Nadezhda-n, 2023-01-26:
[image]
Micro_SD_Card on the bottom. 
I guess it needs to be moved?

rusefillc, 2023-01-26:
Ohhh let's do nothing for now that would be safest?

Nadezhda-n, 2023-01-26:
https://github.com/rusefi/alphax-4chan/tree/New_rev include this

ElDominio, 2023-01-27:
> Ohhh let's do nothing for now that would be safest?

Will MAP sensors still bruise easily?

rusefillc, 2023-01-29:
> Ohhh let's do nothing for now that would be safest?

@Nadezhda-n can we apply the copper change (connected NC pads and vias) but not the huge whole? i.e. can we only make progress which does not require SD card to be moved?

rusefillc, 2023-01-30:
2chan has another improvement and via to reinforce pads

[image]

Nadezhda-n, 2023-01-31:
[image]

closed 2023-01-31 by rusefillc