rusefi/alphax-4chan #67: CKP pad forever grounded

ElDominio, 2022-10-08:
This photo is probing CKP pin directly

[image]

This photo is probing CKP pad
[image]

This photo is probing CMP pad
[image]

rusefillc, 2023-03-14:
@ElDominio please elaborate

which revisions
why do we care
how can I test
what is desired

ElDominio, 2023-03-14:
This looks like a Lancer, lancer works fine lately except for hall noise but not related to this

closed 2023-03-14 by ElDominio