rusefi/alphax-4chan #68: VVT target seems to be locked at 25 degrees for intake, settings min/max duty to 0 makes no difference

ElDominio, 2022-10-08:
https://rusefi.com/online/view.php?log=1073

In this log, the VVT solenoid was unplugged and angle was 0. The instant vvt is connected, VVT angle goes to ~25 and nothing I change has any effect, only disconnecting it.

ElDominio, 2022-10-09:
[image]

Oscope looks like this

rusefillc, 2023-01-30:
@ElDominio what's the status? this looks like firmware issue?

ElDominio, 2023-01-30:
Actually idk, this ecu is from the Chinese seller and he kept using it in the NOS output.

rusefillc, 2023-03-13:
@ElDominio what is this ticket about? Please look at it through me eyes.

ElDominio, 2023-03-15:
This was probably due to bad wiring, closed.

closed 2023-03-15 by ElDominio