rusefi/alphax-4chan #71: parts shortage today

rusefillc, 2022-10-26:
Marsel is looking to order a batch of 4chan but BOM shortages

R4,R49 | 4D03WGF1001T5E | R0603-4 | 4D03WGF1001T5E C425239

U103 | AP2331W-7 | SOT-23 | AP2331W-7 C460346

F5,F6 | Fuse-5A-1206 | SMD1206 | S1206-F-5.0A C553924

rusefillc, 2022-10-26:
@andreika-git can I simply remove U103 from alphax-4chan BOM?

rusefillc, 2022-10-26:
https://github.com/andreika-git/hellen-one/wiki/JLC-fabrication-component-substitution-options is the relevant page to update

rusefillc, 2022-10-26:
R0603-4 extended part I know @Qwerty-OFF just wants to replace with four individual 0402?

andreika-git, 2022-10-26:
> R4,R49 | 4D03WGF1001T5E | R0603-4 | 4D03WGF1001T5E C425239

[image]

andreika-git, 2022-10-26:
> U103 | AP2331W-7 | SOT-23 | AP2331W-7 C460346

AP2331SA-7 C500764

andreika-git, 2022-10-26:
> F5,F6 | Fuse-5A-1206 | SMD1206 | S1206-F-5.0A C553924

JFC1206-1500FS C136351

closed 2023-03-13 by rusefillc