rusefi/alphax-4chan #75: ETB driver overheating

ElDominio, 2022-11-15:
In two cars already, when the car is keyed on ETB driver keeps having a very hard time to control ETB position until it overheats and shutdown. Any way to improve cooling?

Temperature tested by touching the driver on key on

rusefillc, 2022-11-15:
@ElDominio what revisions?

We had a change between B and C open question if that was improvement or degradation. @mck1117 @Qwerty-OFF do we have potential for improvement?

rev A
[image]

rev B
[image]

rev C
[image]

ElDominio, 2022-11-15:
I think when JLC assembly started, they only solder pads, not chip to board

mck1117, 2022-11-15:
they've always soldered the thermal pad underneath, and we've never had a problem with the 9201 on proteus or MRE

ElDominio, 2022-11-18:
Tested between revB and revC, about 25F difference in temperatures

revB has no holes, revC does

mck1117, 2022-11-18:
> Tested between revB and revC, about 25F difference in temperatures (revB is cooler)

which is worse? B = C + 25 or C = B + 25?

ElDominio, 2022-11-18:
> > Tested between revB and revC, about 25F difference in temperatures (revB is cooler)
> 
> which is worse? B = C + 25 or C = B + 25?

I said revB is colder lol, meaning C = B + 25

mck1117, 2022-11-18:
head empty no thoughts

ElDominio, 2022-11-18:
https://www.ebay.com/itm/185608066289?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=x791hh4IT1u&sssrc=2349624&ssuid=spRb3_QBRRK&var=&widget_ver=artemis&media=COPY ordered this too

mck1117, 2022-11-18:
Those won't do very much. The die is glued or soldered to the copper slug on the bottom, so you're going through a few mm of relatively thermally unconductive epoxy on the top. Can you try reflowing that chip with some more solder?

ElDominio, 2022-11-18:
I had the guy that does pro reflowing do them today and there was basically no change. Wouldn't the holes make it have less surface area for conducting heat so make it heat up quicker?

rusefillc, 2022-11-18:
@ElDominio i have an IR camera do you have an IR camera?

@ElDominio does it look like it's specific kind/model/vehicle ETB giving us problems or any ETB in general? I wonder what and how can I reproduce in NJ

ElDominio, 2022-11-18:
https://www.ebay.com/itm/174881539248?hash=item28b7c084b0:g:7wMAAOSwwD5frmfY&amdata=enc%3AAQAHAAAAoHhLNzeWt4g8aUDswxNKG4UiYoEY0YnC9DmtHNLpjytCGVz%2FJCz7W1aeStCqqfUdm8Kmp0nbn2EX%2BwKPIath4J%2Bt1Xfh3tKeEi%2FPfhQp1owPFBW0vYIgaEg7CBiEN5Nx2GgTOqW%2BD9kC5dTWB3sfa8B0QwErreynAhfov5SAuBvK1fpPfBYnb%2FXxqiEXzzLojYhR2jH0XqGNxeAUa18kwwk%3D%7Ctkp%3ABk9SR6jH8biRYQ

That's what I measured with.

Right now I have it on 4 different setups and they all overheat. 
Where should we be grounding the ECU? I always tell my kids to ground directly to battery but might that be causing issues?

rusefillc, 2022-11-19:
@ElDominio says this happens while engine is not running just ignition is ON

I am starting to wonder if that's tune / control strategy issue? Do we push these shut with 100% duty cycle or something? Need logs and tune.

rusefillc, 2022-11-21:
Linking just in case https://github.com/rusefi/rusefi/issues/4816

ElDominio, 2022-11-22:
This is a log of an ETB converted car we have at the shop, looks for resting position and fails a bit into the log: -> https://rusefi.com/online/view.php?log=1103

Same car trying to autocalibrate, I have tried a myriad of values on feed forward  and they don't make a difference -> https://rusefi.com/online/view.php?log=1104

If any videos are needed let me know; tune is tied to logs

ElDominio, 2022-11-22:
This is another ETB conversion with a fifth gen Camaro throttle, 2012 throttle. Catching this car failing is difficult since it _works_ most of the time, but it can someotimes give up randomly since it runs extremely hot.

Engine off key on -> https://rusefi.com/online/view.php?log=1105

running PID autocalibrate -> https://rusefi.com/online/view.php?log=1106 changed feed forward to be less positive in second run, you can still see it not getting exactly to 50

rusefillc, 2022-11-23:
@ElDominio can you please post tune + log of Ray's blue car, ignition on engine off?

My point is -  car at the shop I see crazy oscillation of duty cycle, I am comparing with nissan duty cycle chart and things are MUCH much smoother. Maybe energy magic happens when you change duty cycle that often that much?

shop car note crazy red line on top. Within two seconds it's constantly going -40% to +60%

[image]

now a log of my nissan sitting with engine off. Note how small is ETB Duty change when I am not touching pedal

[image]

@mck1117 how reasonable is my "energy"/total amount of change concern? Do we need some sort of volatility metric?

rusefillc, 2022-11-23:
Between VBat drop and ETB Duty volatility I am _guessing_ that ETB volatility is a much bigger concern.

Step one: need someone smart to voice their opinion on ETB Duty volatility and that's pretty much @mck1117 maybe @Qwerty-OFF @dron0gus @andreika-git 

if/when we agree to blame PID settings next question would be why has auto calibrate failed. Is that a defect in code or could any power concerns be causing it.

Cherry on the case https://github.com/rusefi/rusefi/issues/4816 would that make a pattern of auto-calibrate being poor specifically on 4chan?! looks like next step would be to auto-calibrate same exact ETB on two different boards of different design. Let's see if @mck1117 would ask me to ship my test rig to him?

ElDominio, 2022-11-24:
Question: couldn't different operating states be defined for the throttle control system? 

I've seen some ECUs that jitter for a bit when changing position, then basically "lock in" and stop jittering (LinkECU principally). I'd assume that have different duty limits for different states.

That's also how an ancient throttle controller I had made worked and I would say it worked pretty well, but that's anecdotal evidence. https://github.com/ElDominio/ETC_Control

rusefillc, 2022-11-24:
Answer: https://en.wikipedia.org/wiki/Moon_landing tells us that a lot of things are possible with proper resources.

Yes some (many? all?) h-bridges have "hold" state and it could be relatively easy to implement. But proper approach is first to identify the problem and then start addressing it.

I see very different behavior between your tune and my tune. I believe that first step is to explain it. If the problem is only with calibration we shall focus on calibration

Realistically next step is for me to reproduce the issue in NJ. What _exact_ part number ETB do we have problems with?

Another opinion: first we have to explain https://github.com/rusefi/rusefi/issues/4816

mck1117, 2022-11-24:
> Realistically next step is for me to reproduce the issue in NJ. What exact part number ETB do we have problems with?

Do we want to reproduce this instead in WA where I have the better test equipment to actually measure what’s going on?

mck1117, 2022-11-24:
Also, I don’t know what “hold state” means on an h bridge, but whatever it is, the tle9201 doesn’t have it.

rusefillc, 2022-11-24:
Best option is both WA and NJ :) @ElDominio we need exact part number to start acquiring matching stuff

ElDominio, 2022-11-24:
Bosch 99760511501

And

12629992

rusefillc, 2022-11-24:
That's porsche fancy?! How come?

 and LS less fancy :)

Let me cross-test same ETBs between 4chan and Proteus

rusefillc, 2022-11-25:
@ElDominio my bigger concern is not ETBs but connectors. 

GM seems easiest and I have some VAG and Nissan pigtails for sure. 

@mck1117 do you have a 4chan in WA or is it a 2chan you have?

rusefillc, 2022-11-26:
@ElDominio I have one or two logs and four part numbers for different ETBs. Do you see my problem?

ElDominio, 2022-11-27:
I offered to buy them so you can do it on your end, I was asked for logs and part numbers and I provided them lol

rusefillc, 2022-11-27:
@ElDominio please clarify which log is which ETB, which ETB is most problematic? Looks like GM is not horrible and we better start with a horrible one.

@ElDominio please confirm that you can source pigtails not just ETBs.

ElDominio, 2022-11-27:
I have the same pigtail problem but if i can source a pigtail + etb i will ship

rusefillc, 2022-11-27:
@ElDominio please clarify which log is which ETB, which ETB is most problematic? Looks like GM is not horrible and we better start with a horrible one.

rusefillc, 2022-11-27:
#4832 has convinced me that we have a calibration error here with potential for smarter error detection

I guess that #4833 is the next step

ElDominio, 2022-11-27:
> @ElDominio please clarify which log is which ETB, which ETB is most problematic? Looks like GM is not horrible and we better start with a horrible one.

1103  and 1104 are the most problematic, Bosch throttle body log.

Correct, GM is not horrible. Takes sometimes about a half hour to fail. Bosch fails almost instantly

rusefillc, 2022-11-29:
@ElDominio my Nissan logs are at https://rusefi.com/forum/viewtopic.php?p=46623#p46623 and https://rusefi.com/forum/viewtopic.php?p=46624#p46624

rusefillc, 2022-11-30:
over the phone discussion #79

rusefillc, 2022-11-30:
user error/software error https://github.com/rusefi/rusefi/issues/4832#issuecomment-1331639530

closed 2022-11-30 by rusefillc