rusefi/alphax-4chan #76: Time to remove jumpers

rusefillc, 2022-11-20:
@ElDominio have you ever used any of the jumpers for any reason? can we remove all or replace with test vias if we ever need space on this PCB? if we want to make the PCB simpler?

ElDominio, 2022-11-20:
I didn't know what jumpers?? It has jumpers? Where?

Crank/Cam logic analyzer should be vias and not pads. Little holes i can solder pins to if necessary

rusefillc, 2022-11-21:
@ElDominio these jumpers between connectors and components

[image]

ElDominio, 2022-11-21:
fine to remove

rusefillc, 2023-03-15:
Some were removed to address mounting point issue. Technically theoretically sometimes having at least a via help, having complete jumper might help even more.

We can always remove individual ones if that helps with tracing or else. Replacing with a via keeps at least of some over-engineering.

rusefillc, 2023-03-16:
Actually closing to keep things simper

closed 2023-03-16 by rusefillc