rusefi/alphax-4chan #78: Can we change to USB-C connectors on board? PN: C2764612

ElDominio, 2022-11-28:
Change USB connector to C2764612 , USB-C type since I have too many USB-C cables laying around

rusefillc, 2022-11-28:
https://jlcpcb.com/partdetail/XkbConnectivity-U262_061N4BVC11/C2764612

Qwerty-OFF, 2023-03-15:
https://github.com/rusefi/alphax-2chan/issues/68#issuecomment-1336463362

ElDominio, 2023-03-15:
https://jlcpcb.com/partdetail/Dealon-USB_TYPE_C018/C2927038

This is compatible?

rusefillc, 2023-03-15:
@Qwerty-OFF do you see proper connectors at JLC?

@ElDominio do you want USB-C to
* pay
* wait an extra 1 to more weeks
For global sourcing not jlc library?

mck1117, 2023-03-15:
> https://jlcpcb.com/partdetail/Dealon-USB_TYPE_C018/C2927038
> 
> This is compatible?

be careful with the design rules limits on that, the pins are very small

Qwerty-OFF, 2023-03-15:
[image]

rusefillc, 2023-03-16:
Wow funny timing

[image]

closed 2023-03-18 by Qwerty-OFF