rusefi/alphax-4chan #81: Test points for trigger logic analyzer

mck1117, 2022-12-14:
Add test points for 0.1" headers to more easily probe cam/crank inputs (digital and VR).

rusefillc, 2022-12-17:
@mck1117 do you mean that SMD pads are not user friendly? i.e. "more easily" than what?

rusefillc, 2022-12-17:
vias should be possible once 4chan migrates to megamodule

mck1117, 2022-12-17:
> @mck1117 do you mean that SMD pads are not user friendly? i.e. "more easily" than what?

SMD pads require soldering or a human holding something on there. Most logic analyzers use 0.1" header pin wires, so adding some 0.1" holes to solder a header would optimize scenario of install/remove logic analyzer.

rusefillc, 2023-01-22:
https://github.com/rusefi/alphax-4chan/issues/87 would kind of help

closed 2023-03-15 by rusefillc