rusefi/alphax-4chan #82: Two mounting points are grounding stuff

rusefillc, 2022-12-15:
[image]

[image]

rusefillc, 2022-12-15:
@ElDominio how come this is not an issue for you?

https://www.instagram.com/marsel_zx/ reports actual injector short

rusefillc, 2022-12-15:
i suggest removing the jumpers which are causing the issue and adding proper bolt mounting points

ElDominio, 2022-12-15:
Because i tape the bottom of the case because this happened on the old 4chans lol

rusefillc, 2022-12-15:
[image]

Nadezhda-n, 2023-01-26:
**_some_** 
are there sizes?

rusefillc, 2023-01-26:
@Nadezhda-n let's remove 7 left and 7 right to behind with, sizes on Sunday

Nadezhda-n, 2023-01-26:
> @Nadezhda-n let's remove 7 left and 7 right to behind with, sizes on Sunday

wait for sunday

Nadezhda-n, 2023-01-30:
@rusefillc

rusefillc, 2023-01-30:
диаметр d=6.2mm
расстояние между центрами примернр 86.7мм

мы там не делаем крепёжных отверстий - просто делаем гладкую плату без голой меди

Nadezhda-n, 2023-01-31:
> диаметр d=6.2mm расстояние между центрами примернр 86.7мм
> 
> мы там не делаем крепёжных отверстий - просто делаем гладкую плату без голой меди

а высота какая?
Плата ровно по центру корпуса?

rusefillc, 2023-01-31:
По центру

closed 2023-01-31 by Nadezhda-n