rusefi/alphax-4chan #90: Aux connector

rusefillc, 2023-01-30:
* GND
* 5v
* analog inputs

Same physical connector as WBO on https://github.com/rusefi/hellen121nissan/ PCB
[image]

ElDominio, 2023-03-15:
12v controllable means

When i activate in firmware, produces 12v on the pin, not gnd

rusefillc, 2023-03-16:
@Qwerty-OFF @ElDominio should we be freaking out with SMD connector like on the picture? shall we go through-hole for mechanical safety?

ElDominio, 2023-03-16:
I'd say SMD is ok? If users rip it off means bad wiring, I would think

mck1117, 2023-03-16:
if you have the option, why would you ever use SMD for a user-interfacing connector? That's asking for warranty claims, man.

rusefillc, 2023-03-16:
@Qwerty-OFF we voted through-hole.

ElDominio, 2023-03-16:
I know very little about hardware which is why i used question marks. If THT is better, use THT

rusefillc, 2023-03-18:
Same ignition as on #96 but two more injection channels for aux connector

Qwerty-OFF, 2023-03-19:
[image]

closed 2023-03-19 by Qwerty-OFF