rusefi/alphax-4chan #97: Lua output pin error

ElDominio, 2023-02-10:
[image]

[image]

What am i doing wrong?

lua script at https://gist.github.com/ElDominio/cf124fe3b665f734d4afa2c5844dee22

rusefillc, 2023-02-10:
I've attempted reproducing but I've failed

if you reboot ECU does it go red right away?

Please provide a URL to full tune

ElDominio, 2023-02-13:
[image]

doesn't go away on reboot

https://rusefi.com/online/view.php?msq=1396

ElDominio, 2023-02-13:
FIXED:

[image]

Used output1 and it stopped complaining (changed Lua code to use Output1)

rusefillc, 2023-02-13:
So weird :(

Not fixed but worked around.