rusefi/alphax-4chan #100: ETB behavior on 4chan

ElDominio, 2023-02-13:
So I have had differing experiences with Hellen-based hardware on hellen154 vs 4chan

On Hyundai 154, I can use autotune just fine, and the PID settings are usually 
P = 0xx.xx
I = 0xxx.xx
D = 0.xxxx

On 4chan
P = 0x.xx (usually around 0.8-2)
I = 0xx.xx (usually around 10-15)
D = 0.0xxx (usually around 0.02-0.09)
Using autotune drives it too crazy, and usually overheats. Using settings by hand usually makes it works quite well.

rusefillc, 2023-02-13:
@Qwerty-OFF any ideas why would Hellen-based Hyundai 154 use ETB PID settings matching Proteus while Hellen-based would be pretty different, with PID auto-tune working on Proteus and Hyundai 154 but failing on 4chan?

mck1117, 2023-02-13:
What sort of capacitors does hyundai board use near the ETB chip? Ceramic or electrolytic?

Proteus -> electrolytic

4chan -> ceramic

rusefillc, 2023-02-13:
old hyundai https://raw.githubusercontent.com/wiki/rusefi/rusefi/Hardware/Hellen/hellen154hyundai-rev-b.jpg ceramic?

new hyundai I do not have any photo :( @ElDominio ?

8chan ceramic

Qwerty-OFF, 2023-02-27:
[image]

rusefillc, 2023-02-28:
@mck1117 adding electrolytic did not change anything :(

[image]

rusefillc, 2023-03-16:
@dron0gus says

[image]

[image]

rusefillc, 2023-03-16:
that did not help :(

[image]

[image]

rusefillc, 2023-03-16:
@mck1117 @andreika-git I am out of ideas. Sounds like we need an EE person to touch this? Would any sort of logs help do we even log useful stuff?

#109 created because why not

rusefillc, 2023-03-16:
https://github.com/rusefi/alphax-4chan/issues/110 another crazy one

mck1117, 2023-03-16:
I don't think there are any useful logs to have here that aren't an oscilloscope.

mck1117, 2023-03-16:
actually, can you take a maxed out TS rate log of auto calibrate both in the GOOD and BAD scenarios, for clear side-by-side comparison?

rusefillc, 2023-03-21:
@mck1117 poke about above good vs bad

andreika-git, 2023-03-24:
The 4chan log has "OBD_TPS1_Correlation" error, which means the problem is with TPS1 (hardware or wrong settings?) - that's why PID autotune fails.

mck1117, 2023-03-24:
bad left, good right

[image]

In the good case, it looks more sinusoidal, and in the bad case, it looks much steeper on opening than closing.

In the bad case there are also some weird "runt" cycles like this:
[image]

In the good case it's much more "stable"

mck1117, 2023-03-24:
> The 4chan log has "OBD_TPS1_Correlation" error, which means the problem is with TPS1 (hardware or wrong settings?) - that's why PID autotune fails.
> 
> The OBD_TPS1_Correlation error appears when one of TPS sensors (TPS1 or TPS2) has an invalid signal. As a test, we can disable the TPS2 and try in 'ignoreSecond' mode.

That warning isn't active at the time autotune is running. That's not the problem.

mck1117, 2023-03-24:
> that's why PID autotune fails.

The logs show the firmware is doing fine, and doing the same thing in both cases. It also looks like something is electrically or mechanically wrong: see the weird TPS waveform on the "bad" log.

mck1117, 2023-03-24:
Wait, alphax 4chan is using pin C6 for PWM.

`stm32_common.cpp` says

[image]

But `mcuconf_common_f4_f7.h` says

[image]

4CHAN IS USING SOFTWARE PWM FOR ETB

closed 2023-03-27 by rusefillc

mck1117, 2023-03-27:
> Damn that's a shameful issue :(

this should be a lesson for “electrical weirdness -> get out a scope and look at it instead of shooting in the dark for a month”

ElDominio, 2023-03-29:
@rusefillc was this tested before closing?

rusefillc, 2023-03-29:
> @rusefillc was this tested before closing?

yes. my bad not posting happy artifacts

and this justifies a new release actually!

ElDominio, 2023-03-29:
Tested today, way, way better results! :D