rusefi/alphax-4chan #102: rev E cap mod

rusefillc, 2023-02-22:
How do we hand modify add caps for hall digital noise drama?

https://github.com/rusefi/alphax-4chan/issues/72

pins C3 and C7

actually it's mostly C7 which has issue. Is it same VR cap or different cap to improve mostly C7 digital hall input?

[image]

ElDominio, 2023-02-22:
Question, does changing to E5 help at all?

rusefillc, 2023-02-22:
Question: is there a difference between C3 and C7?

Qwerty-OFF, 2023-02-22:
C7 is traditional OpAmp input through Input module
C3 connected to VR discrete and Schmitt trigger
Try to add 1n capcitor:
[image]
[image]
And remove this:
[image]

rusefillc, 2023-02-23:
@timbelom please attempt two above mentioned mods on 4 chan rev E we have

* take 0805 cap marked 1n
* remove resistor as described

rusefillc, 2023-02-23:
@Qwerty-OFF посоветуй плиз куда кидать на землю конденсатор рядом с R22 тоже картинкой?

Qwerty-OFF, 2023-02-24:
[image]
[image]
Лучше использовать SMD

rusefillc, 2023-02-26:
@Qwerty-OFF my bad we have rev F not rev E, need same instructions for rev F with latest VR

rusefillc, 2023-02-26:
@Qwerty-OFF rev C does not have D57 is there a chance to fix rev C? 

overall at the moment we have rev F and rev C to modify

rusefillc, 2023-02-27:
1nf
[image]

rusefillc, 2023-02-27:
Actually @Qwerty-OFF says to remove R753 to disconnect VR just for a test

rusefillc, 2023-02-27:
@ElDominio this board is on the way to you. Please test and share results :)

rusefillc, 2023-04-01:
@Qwerty-OFF we have positive test results! we need this mod on rev G

https://github.com/rusefi/rusefi/issues/5211#issuecomment-1493075254

Qwerty-OFF, 2023-04-02:
[image]

closed 2023-04-02 by rusefillc