rusefi/alphax-4chan #104: rev F noise on C7

rusefillc, 2023-03-08:
[image]

https://rusefi.com/online/view.php?msq=1416&dialog=triggerConfiguration cam uses C7

Qwerty-OFF, 2023-03-08:
Отличия от обычного входа CAM:
- подтяжка настраиваемая;
- подтяжка выполнена к +5VAS (TLS115), а не +5VA как обычно.

Давай попробуем сделать обычную подтяжку, для этого нужно отпаять R44 и поставить 4.7 кОм как на картинке
[image]

Следующий эксперимент с конденсатором C200, сейчас там 1 нФ, можно поставить 10 нФ и посмотреть что случится.

rusefillc, 2023-03-08:
@ElDominio I have mod instructions I need stuff to mod please :)

rusefillc, 2023-03-15:
* on two boards:
@Timbelom we have 5.6K through-hole resistors for sure, could be easier than 4.7K SMD
but we can try 4.7K maybe 1206 mounted vertically and a small wire? 

* let's make one board with C200 replaced, one without C200 replaced

https://rusefi.com/docs/ibom/alphax_4ch-f-ibom.html
[image]

rusefillc, 2023-04-01:
@Qwerty-OFF we have positive test results! we need the mod on rev G. Looks like C200 part of the mod is not needed

https://github.com/rusefi/rusefi/issues/5211#issuecomment-1493075254

Qwerty-OFF, 2023-04-02:
[image]

rusefillc, 2023-04-02:
@andreika-git what are your thoughts? what change could have changed what why?

ElDominio, 2023-04-12:
@Qwerty-OFF question, would removing R44 and adding the 4.7k Resistor be an acceptable fix?

Qwerty-OFF, 2023-04-12:
@ElDominio yes
[image]

ElDominio, 2023-04-13:
But I also have to add a capacitor on D57 right?

rusefillc, 2023-04-13:
> D57

@ElDominio your comment seems to be first mentioning of D57 on this ticket? Please provide some context?

ElDominio, 2023-04-13:
I am trying to mod a RevC to work with 4G63

https://github.com/rusefi/alphax-4chan/issues/102#issuecomment-1446834216

Shows a capactior added to D57, but D57 does not exist on revC. I'm trying to figure out an equivalent position to place said capacitor to see if I can make it work.