rusefi/alphax-4chan #106: We need space for much more got glue around MAP

rusefillc, 2023-03-15:
* We need space for much more got glue around MAP
* can we make a couple of 2mm vias on sides of MAP for hot glue to attach to holes?

Pictured rev F

[image]

ElDominio, 2023-03-15:
I think more glue is not the solution

Move MAP location nearer to the middle of the board or close to connector, to evade damage during shipping

[image]

Like so
[image]

mck1117, 2023-03-15:
> I think more glue is not the solution

Besides, no amount of glue will save you. The pins/PCB are stiffer than the glue, so any deflection will result in stress in the pins, not the glue. The pins will snap off, then the glue will bear the load only once that's happened. You need either a stiff urethane glue or epoxy, mechanical connection (ie, bolt), or move the sensor.

rusefillc, 2023-03-15:
@ElDominio says

Move MAP location nearer to the middle of the board or close to connector, to evade damage during shipping

[image]

Like so
[image]

rusefillc, 2023-03-15:
> move the sensor.

I agree, location on the edge seems to be the main issue

Qwerty-OFF, 2023-04-02:
[image]

rusefillc, 2023-04-02:
@Qwerty-OFF still on the edge :(

rusefillc, 2023-04-02:
@Qwerty-OFF can we move to 

[image]

so that it's NOT on the edge of the board?

Qwerty-OFF, 2023-04-02:
[image]

closed 2023-04-02 by Qwerty-OFF