rusefi/alphax-4chan #110: ETB another idea

rusefillc, 2023-03-16:
Do we have space for electrolytic capacitor to be much close to MRE/proteus with simpler modifications?

Qwerty-OFF, 2023-03-18:
[image]

[image]

andreika-git, 2023-03-22:
Are we sure that the pcb file is updated with the new cap?
[image]

Qwerty-OFF, 2023-04-01:
[image]

closed 2023-04-01 by Qwerty-OFF