rusefi/alphax-4chan #115: Suspect VR threshold is affecting ALL kinds of timing

ElDominio, 2023-03-31:
I have a car on the dyno with this curve on the VR threshold
[image]

If I change to this:
[image]

The powerband loses power all across the board (in the tune of 3hp on a 100hp car)

This leads me to believe VR threshold settings affect timing.

mck1117, 2023-03-31:
does this board have the added caps to the vr module?

ElDominio, 2023-03-31:
Does not

mck1117, 2023-03-31:
Then this is expected. You need the caps.

closed 2023-03-31 by mck1117

ElDominio, 2023-04-01:
Good to know heh thanks

rusefillc, 2023-04-01:
https://github.com/rusefi/rusefi/wiki/AlphaX-4chan#changelog says rev F has VR fix