rusefi/alphax-4chan #118: rev G 4g status

rusefillc, 2023-04-02:
Between firmware progress and hardware mods we got happy 4g running rev F!

* More than one EE is asking questions how does configurable VR/Hall even work https://github.com/rusefi/alphax-4chan/issues/117
* In light of above, we add safe capacitors and provision footprints for optional pull-ups see [102](https://github.com/rusefi/alphax-4chan/issues/102#issuecomment-1493395593)

rusefillc, 2023-04-17:
* usb-c took DFU firmware
* blue LED blinks

rusefillc, 2023-04-17:
* ETB moves with auto-calibrate works
* both TPS input sensors work
* injectors click in self-stim

rusefillc, 2023-04-18:
board No 2

[image]

rusefillc, 2023-04-18:
board No 3

[image]

rusefillc, 2023-04-18:
rev F just for reference. same exact firmware same exact testing apparatus

[image]

Qwerty-OFF, 2023-04-18:
PPS1:
[image]

rusefillc, 2023-04-18:
mux1,2 shows zero while R67 shows 2.6v

rusefillc, 2023-05-10:
@ElDominio says CAN bus works good on this 4chan rev G / mm-0.3