rusefi/alphax-4chan #121: 4g experiments without touching VR pins

rusefillc, 2023-04-15:
We shall try to run 4g on rev F and rev G while NOT using C2,C3,E5,E6

meaning that we have C7 E3 E4

rusefillc, 2023-04-15:
@Qwerty-OFF E3 and E4 are they hall-compatible on either F or G?

rusefillc, 2023-04-15:
E3 and E4 have 100nF caps which would need to be removed or better changed to 1-10nF

ElDominio, 2023-04-15:
I have confirmed a simple fix on revC and revE

On revC, I did the same as was said by @Qwerty-OFF , adding 1nf caps to IN_CRANK and a 4.6k resistor to R7 jumper hole to R40 pad (removing R44 in the process)
[image]

still revC, added 1nf caps to IN_CRANK as shown here
[image]

and to D57 (on revF schematic) pad on R24
[image]

This allowed me to read RPM perfectly on 4G93 on a revC
[image]

ElDominio, 2023-04-15:
On revE, I **_ONLY_** added a 1nf cap from bottom pad of R24 to ground
[image]

And also reads perfectly
[perfect4reading4gonlyonecap.zip](https://github.com/rusefi/alphax-4chan/files/11240484/perfect4reading4gonlyonecap.zip)

This revE has stock R44 resistor, no resistors removed to disable VR, no external resistors, and no cap on IN_CRANK. Still works great.
[image]

ElDominio, 2023-04-16:
Fun fact: revE with 1nf cap ALSO works on VR car, specifically 24-1 cam speed Honda B20, with settings from unmodified ECU.

closed 2023-12-05 by ElDominio