rusefi/alphax-4chan #125: test rev G default configuration with R4 removed board number 0 same board different mod

rusefillc, 2023-04-20:


ElDominio, 2023-04-22:
C4 is VR-, did you mean c3?

Qwerty-OFF, 2023-04-22:
С4 is IN_2STEP
[image]

ElDominio, 2023-04-22:
Feed one of the signals through 2step?

Qwerty-OFF, 2023-04-22:
Yes, use it as hall input

ElDominio, 2023-04-27:
I can't choose C4 as input 
[image]

ElDominio, 2023-04-27:
Hall test with revg
4B11 engine 36-2-1
working fine with revC

tune
[tunefor4b11revg.zip](https://github.com/rusefi/alphax-4chan/files/11348693/tunefor4b11revg.zip)

board0:
[revGhallv0.zip](https://github.com/rusefi/alphax-4chan/files/11348694/revGhallv0.zip)

board1:
[revGhallv1.zip](https://github.com/rusefi/alphax-4chan/files/11348695/revGhallv1.zip)

board2:
[revGhallv2.zip](https://github.com/rusefi/alphax-4chan/files/11348696/revGhallv2.zip)

ElDominio, 2023-04-28:
tl;dr all good, did not try revving

rusefillc, 2023-04-28:
@ElDominio why did not try revving?

ElDominio, 2023-04-28:
On Hall i did not, but I can test

ElDominio, 2023-12-05:
RevG works with Hall/VR

closed 2023-12-05 by ElDominio