rusefi/alphax-4chan #126: test rev G test VR

rusefillc, 2023-04-20:
@ElDominio need to run this board as VR C2+C3

rusefillc, 2023-04-20:
board 1 has C17 removed

[image]

rusefillc, 2023-04-20:
board 2 has C20 added over default
[image]

ElDominio, 2023-04-26:
https://rusefi.com/online/view.php?log=1228

With resistors removed reads VR, without removing them, it did not read. Log is with resistors removed.

This car has run on revB/revC/revD so it seems to be a happy simple VR, pends testing on a car with more annoying VR

rusefillc, 2023-04-27:
@ElDominio 
* fact 1: any rev board is useless in the as-soldered-on-factory resistor configuration
* fact 2: as of now I have not removed any resistor from any of the five rev G boards in existence
* conclusion: @ElDominio to remove R from any test board for any test

ElDominio, 2023-04-27:
board 2 log, r4 and r23 removed:
[revGtestVRver2.zip](https://github.com/rusefi/alphax-4chan/files/11347674/revGtestVRver2.zip)

board 1 log, r4 and r23 removed:
[revGtestVRver1.zip](https://github.com/rusefi/alphax-4chan/files/11347711/revGtestVRver1.zip)

board 0 log, r4 and r23 removed: 
[revGtestVRver0.zip](https://github.com/rusefi/alphax-4chan/files/11347787/revGtestVRver0.zip)

all tests done on Toyota 1NZ 36-2 Crank, 4-1 cam, dual VR
tune file
[tuneforyaris.zip](https://github.com/rusefi/alphax-4chan/files/11347805/tuneforyaris.zip)

rusefillc, 2023-04-27:
@ElDominio tl,dr?

ElDominio, 2023-04-27:
all broken, can see sync failure on log

ElDominio, 2023-04-28:
This was tested on problematic VR from Toyota Yaris 1NZ

All boards could not rev past idle due to sync issues, the car works much better on revF

Easy VR from Honda B20 work fine on board 0

Qwerty-OFF, 2023-04-28:
Please remove R22, R35, R1126, R1135 on board 1 (without C17, C20) and check VR C2+C3 again

rusefillc, 2023-04-28:
@ElDominio https://rusefi.com/docs/ibom/alphax_4ch-g-ibom.html

ElDominio, 2023-04-29:
I think it is worse now
[moddedwithoutalotofresistorsboardv1revG.zip](https://github.com/rusefi/alphax-4chan/files/11360218/moddedwithoutalotofresistorsboardv1revG.zip)

rusefillc, 2023-04-29:
@ElDominio please add a photo with all mods?

ElDominio, 2023-04-29:
![IMG_20230429_192301.jpg](https://user-im[image]

17-02864297-7550-47fe-8a32-603438daf84f.jpg)

[image]

rusefillc, 2023-04-29:
[image]
[image]

ElDominio, 2023-07-23:
Can this and revG tickets be closed yet? I don't see why not?

closed 2023-07-23 by rusefillc