rusefi/alphax-4chan #128: VR VVT issues

ElDominio, 2023-04-26:
In this log i change VR threshold for cam input, and the cam angle changes, I don't think this is right, started at 0.3 and kept adding .1 until i got to 1

https://rusefi.com/online/view.php?log=1229

you can see VVT angle change as i change VR threshold

mck1117, 2023-04-26:
Is this an ECU with fixed capacitors?

ElDominio, 2023-04-26:
Yep, revF

mck1117, 2023-04-26:
where's the tune

mck1117, 2023-04-26:
also, have a trigger scope?

ElDominio, 2023-04-26:
Tune is attached to log on rusefi online ;_;
https://rusefi.com/online/view.php?msq=1459 here's the link anyways

no trigger scope cause I don't have the car, should have it on Thursday

ElDominio, 2023-04-27:
[revF oscope log.zip](https://github.com/rusefi/alphax-4chan/files/11347615/revF.oscope.log.zip)

i managed to stall it mid log

mck1117, 2023-04-28:
Trigger scope looks fine. Your triggers are configured correctly for what I see on the scope.

Thresholds also look fine, you can set them pretty high once the engine is running. Not sure why you're seeing timing move with threshold level.

Aside: why does the VVT position move with RPM?

ElDominio, 2023-04-28:
Because I targetted different positions with RPM? i guess lol

I set inverted to True and it seems MUCH more stable, as in useable. I'll close this for now since it seems to be okay

closed 2023-04-28 by ElDominio