rusefi/alphax-4chan #129: VR+hall quesiton and suggested fix idea

ElDominio, 2023-04-29:
revF VR+Hall circuit seems to work quite well, only needing a 1nf cap Net-D57 to GND as shown https://github.com/rusefi/alphax-4chan/issues/121#issuecomment-1509942883 here.

Why did the revG circuit change so much? Can we just do revF VR circuit + the previously mentioned capacitor? Removing/adding said capacitor sounds much less painful than the current state

rusefillc, 2023-04-29:
Really question is "is it possible to adjust current rev G pieces to match rev F 100%?"

Qwerty-OFF, 2023-04-29:
For F ~100%
Remove C135, C141, C142 (because in MM the capacitance on "IN_Dx" lines is 2 x 1nf)
Remove C17, C20, C21
https://github.com/rusefi/alphax-4chan/issues/121#issuecomment-1509942883 not needed (already done)
Remove R4, R39
Install R21, R23 if they are removed
This covers everything related to the lines IN_CRANK+, IN_CRANK-, IN_CAM_VR+, IN_CAM_VR-, IN_CAM_HALL, and IN_2STEP.

ElDominio, 2023-04-29:
How about RevF on IN_CRANK+/- and IN_CAM_VR+/-? Without changing IN CAM HALL and IN 2STEP?

ElDominio, 2023-04-29:
> For F ~100% Remove C135, C141, C142 (because in MM the capacitance on "IN_Dx" lines is 2 x 1nf) Remove C17, C20, C21 [#121 (comment)](https://github.com/rusefi/alphax-4chan/issues/121#issuecomment-1509942883) not needed (already done) Remove R4, R39 Install R21, R23 if they are removed This covers everything related to the lines IN_CRANK+, IN_CRANK-, IN_CAM_VR+, IN_CAM_VR-, IN_CAM_HALL, and IN_2STEP.

On what board? V0, V1, V2?

rusefillc, 2023-04-30:
Your choice?

Reminder that v0 has c17 as from factory and one of v1/v2 has c17 removed and moved on another board as C20. All other mods you've done yourself.

rusefillc, 2023-04-30:
https://rusefi.com/docs/ibom/alphax_4ch-g-ibom.html is helpful to locate C17 and C20

ElDominio, 2023-04-30:
> For F ~100%
> Remove C135, C141, C142 (because in MM the capacitance on "IN_Dx" lines is 2 x 1nf)
> Remove C17, C20, C21
> https://github.com/rusefi/alphax-4chan/issues/121#issuecomment-1509942883 not [...]

Question, with the current setup, can IN_2STEP be used as either a Hall input _or_ 2step input? Because I like that and I tested C4 already and it worked

>  your choice?\n\nReminder that v0 has c17 as from factory and one of v1/v2 has c17 removed and moved on another board as C20. All other mods you've done yourself.

I will test this hopefully tomorrow.

rusefillc, 2023-04-30:
@ElDominio  what is 2step input? just a mechanical button. Hall still has pull-up I assume (you can check with a digital multimeter by checking for disconnected wire) thus yes, same hall or just button

ElDominio, 2023-04-30:
> @ElDominio what is 2step input? just a mechanical button? Hall still has pull-up I assume (you can check with a digital multimeter by checking for disconnected wire) thus yes, same hall or just button

Yep, a mechanical button that pulls to ground.
So it would be best to not modify that input and keep it as is, then have ONLY the two VR circuits as they are in revF

rusefillc, 2023-04-30:
> So it would be best to not modify

I have no idea what that means.

a) are you requesting we undo rev G and start from rev F again?
b) are you requesting something else? what is the ask specifically?

ElDominio, 2023-04-30:
When I test I will have a more concrete request, so no action until then

My current request is:
How do i modify IN_CRANK+/- and IN_CAM+/- without touching IN_2STEP in revG board2?

rusefillc, 2023-04-30:
@Qwerty-OFF says

Something still needs to be done with 2step, either as in https://github.com/rusefi/alphax-4chan/issues/129#issuecomment-1528831105 to get 100% rev. f, or install R39 and remove R23 to get a better chance for the HALL or button on the 2_STEP line to work correctly

ElDominio, 2023-05-01:
> For F ~100% Remove C135, C141, C142 (because in MM the capacitance on "IN_Dx" lines is 2 x 1nf) Remove C17, C20, C21 [#121 [...]

what about R1126 and R1135?

ElDominio, 2023-05-01:
[image]

[image]

[image]

[image]

ElDominio, 2023-05-02:
@Qwerty-OFF was I expected to have R22/R33/etc removed or not? All the 1206 resistors?

Qwerty-OFF, 2023-05-02:
R22/R33/etc R1206 should be installed for rev.f comatibility. They are for the IN_CRANK+, IN_CRANK- and IN_CAM_VR+ use as hall sensor input circuits.

Qwerty-OFF, 2023-05-02:
R1126 and R1135 is missing designators on rev. f and rev. g boards (or not?)

ElDominio, 2023-05-02:
Picture on comment 1530226901 updated to explain what was done

[fullymoddedtorevFcomment1528831105ticket129.zip](https://github.com/rusefi/alphax-4chan/files/11378084/fullymoddedtorevFcomment1528831105ticket129.zip)

The car does not seem happy, but it currently runs quite well on a revF (actual revF hardware)

rusefillc, 2023-05-02:
@Qwerty-OFF seems like "100% rev F compatible" somehow is not 100% rev F compatible :(

Qwerty-OFF, 2023-05-02:
I'll think tomorrow about what other differences there are. Is the problem with pins C2+C3?

ElDominio, 2023-05-02:
> I'll think tomorrow about what other differences there are. Is the problem with pins C2+C3?

Good observation, this car uses dual VR C2+c3 AND E5+E6

If E5+E6 is not revF, cam will not read correctly

Qwerty-OFF, 2023-05-03:
CAM VR differences seem to be fixed by https://github.com/rusefi/alphax-4chan/issues/129#issuecomment-1528831105
Remove C211, C216 (or replace with 1nF if possible)
Ensure that R753_2 is not removed from the board

ElDominio, 2023-05-05:
@Qwerty-OFF VR on Toyota 1NZ 36-2  seems to have improved significantly, 
car runs the same as revF (removed c211, c216 replaced with 1nf)
R753_2 was present

logs
regular TS log of engine running:
[crankingrevGmay4.zip](https://github.com/rusefi/alphax-4chan/files/11409612/crankingrevGmay4.zip)

oscope:
[triggeroscoperevgmay4.zip](https://github.com/rusefi/alphax-4chan/files/11409614/triggeroscoperevgmay4.zip)

ref @rusefillc

ElDominio, 2023-05-05:
[image]

ElDominio, 2023-05-05:
I have idled and revved for about 15 minutos now, seems okay

ElDominio, 2023-05-07:
https://docs.google.com/spreadsheets/d/1M2a5KM3GcPGP_YP6OwbdvqBgPWTfrZ21yheXTK44Okg/edit?usp=sharing

Here is a sheet with all current board states, only truly known is board1, will fill out others later

ElDominio, 2023-05-12:
@Qwerty-OFF  i have updated the spreadsheet, and will have the test car for about three more days. Is there any more tests that need to be done?

closed 2023-07-23 by ElDominio