rusefi/alphax-4chan #131: ibom: cannot click on passives above D0203

rusefillc, 2023-05-04:


andreika-git, 2023-05-10:
Cannot reproduce (works on both Chrome and Firefox):

[image]

rusefillc, 2023-05-10:
step 1: click https://rusefi.com/docs/ibom/alphax_4ch-g-ibom.html
step 2: do NOT click "FB" mode
step 3: click above D204 happy
step 3: click above D203 unhappy

andreika-git, 2023-05-10:
Some of 10k/0402 resistors have wrong part numbers (510k instead of 10k). That's why they were duplicated in the BOM - and it could possibly cause this bug of partial selection (those resistors above D203 are 10k). Maybe we should fix the mega-mcu BOM first and then continue to test the 4chan iBom.

rusefillc, 2023-05-10:
@andreika-git стоит ли нам защищаться от некорректных данных, или непросто?

andreika-git, 2023-05-11:
> @andreika-git стоит ли нам защищаться от некорректных данных, или непросто?

https://github.com/andreika-git/hellen-one/issues/293

rusefillc, 2023-05-15:
@Qwerty-OFF please mention relevant 4chan changes just to have a changelog

Qwerty-OFF, 2023-05-15:
Вот у этих компонентов была неправильный PN. Вместо 10k стали 510k.
"10k","R102, R108, R109, R121, R127, R128, R141, R142, R232, R233, R234","R0402","C11616"
В ревизии h все кроме R232 получили правильный PN "C25744". R232 стал DNP (была ошибка в rev. g - https://github.com/rusefi/alphax-4chan/issues/137)

rusefillc, 2023-05-16:
@Qwerty-OFF @andreika-git this ticket is about ibom glitch.

If we have something important going on with five rev G boards in the wild please improve the messaging and making it more obvious what is the level of risk where etc.

andreika-git, 2023-05-16:
https://github.com/andreika-git/hellen-one/issues/295

andreika-git, 2023-05-18:
Fixed by fixing https://github.com/andreika-git/hellen-one/issues/295

closed 2023-05-18 by andreika-git