rusefi/alphax-4chan #132: VVT not working on RevG

ElDominio, 2023-05-05:
VVT seems to not work on revG, testing on Yaris

log: https://rusefi.com/online/view.php?log=1232
tune is linked to log on rusefi online

possible culprit sames as revG PPS?

rusefillc, 2023-05-06:
rev G
[image]

[image]

[image]

[image]

rusefillc, 2023-05-06:
rev F
[image]

rusefillc, 2023-05-06:
@ElDominio yes VVT1 and VVT2 pinout has changed on rev G, now adding "F2 - VVT#1 rev G" and "F4 - VVT#2 rev G" pins

ElDominio, 2023-05-06:
Is there a way to catch any other pins like these?

rusefillc, 2023-05-06:
@ElDominio yes, just go over all pins in the schematics

rusefillc, 2023-05-07:
Also https://rusefi.com/docs/pinouts/hellen/alphax-4chan/ progress has helped since it at least detects pin overlap.

Legacy VVT pins have funny names in latest firmware.

ElDominio, 2023-08-25:
This seems to still be broken; did it break when removing old compatibility?

rusefillc, 2023-08-25:
@ElDominio please clarify why closed?

ElDominio, 2023-08-25:
Did not know if a new ticket was better, asking for processor pins.

rusefillc, 2023-08-25:
https://github.com/rusefi/alphax-4chan/issues/159 not better

rusefillc, 2023-12-03:
@ElDominio too many things bundled together here - random vehicle etc etc

when one suspects a hardware problem on the board we have to start by testing physical output on a bench. clicking a known good test injector using "injector bench" button seems to be the most reliable way

i have a rev G board which passes stim QC testing on both vvt and idle outputs that confirm that firmware is capable of toggling output wire

```
 * Board detected: 4chan rev index 1
                      numOutputs 8
               Setting ADDR 0
               sending line=1 value=1
                      ADC says HIGH 1@0 3.080v
               sending line=1 value=0
                      ADC says LOW 1@1 0.002v
               sending line=1 value=1
               sending line=1 value=0
0/16 A8 - Injector 1 GOOD channel 1
               Setting ADDR 1
               sending line=2 value=1
                      ADC says HIGH 2@0 3.080v
               sending line=2 value=0
                      ADC says LOW 2@1 0.002v
               sending line=2 value=1
               sending line=2 value=0
1/16 B8 - Injector 2 GOOD channel 2
               Setting ADDR 2
               sending line=3 value=1
                      ADC says HIGH 3@0 3.080v
               sending line=3 value=0
                      ADC says LOW 3@1 0.002v
               sending line=3 value=1
               sending line=3 value=0
2/16 D1 - Injector 3 GOOD channel 3
               Setting ADDR 3
               sending line=4 value=1
                      ADC says HIGH 4@0 3.080v
               sending line=4 value=0
                      ADC says LOW 4@1 0.002v
               sending line=4 value=1
               sending line=4 value=0
3/16 E1 - Injector 4 GOOD channel 4
               Setting ADDR 4
               sending line=5 value=1
                      ADC says HIGH 5@0 1.399v
               sending line=5 value=0
                      ADC says LOW 5@1 0.001v
               sending line=5 value=1
               sending line=5 value=0
4/16 F2 - VVT#1 rev G GOOD channel 5
               Setting ADDR 5
               sending line=6 value=1
                      ADC says HIGH 6@0 1.604v
               sending line=6 value=0
                      ADC says LOW 6@1 0.002v
               sending line=6 value=1
               sending line=6 value=0
5/16 F4 - VVT#2 rev G GOOD channel 6
               Setting ADDR 6
               sending line=7 value=1
                      ADC says HIGH 7@0 1.688v
               sending line=7 value=0
                      ADC says LOW 7@1 0.002v
               sending line=7 value=1
               sending line=7 value=0
6/16 F1 - Idle2 GOOD channel 7
               Setting ADDR 7
               sending line=8 value=1
                      ADC says HIGH 8@0 1.720v
               sending line=8 value=0
                      ADC says LOW 8@1 0.001v
               sending line=8 value=1
               sending line=8 value=0
7/16 C8 - Idle GOOD channel 8
```

closed 2023-12-03 by ElDominio