rusefi/alphax-4chan #139: Fix BOM (see warnings.log)

andreika-git, 2023-05-18:
Current revision (rev.I) generates some warnings:

```
* Warning! Footprint mismatch for the part #C269114: R1206 != SMD1206
* Warning! Identical parts C753,C754,C759,C100,C101,C105,C108,C109,C110,C111,C112,C113,C118,C119,C120,C121,C122,C124,C126,C127,C128,C129,C130,C134,C147,C170,C175,C180,C185,C190,C195,C204,C205,C206,C207,C217,C218,C220,C221,C222,C230,C231,C232,C233,C270,C275,C280,C285,C753_2,C754_2,C759_2 and C411,C413,C420,C470 (100n_C0402) have different partnumbers: C1525 and C307331
* Warning! Identical parts VD100,VD101,VD102,VD103 and D18,D19,D23 (BAT54S_SOT-23) have different partnumbers: C47546 and C408389
* Warning! Identical parts R125,R170,R171,R175,R176,R180,R181,R185,R186,R190,R191,R196,R197 and R1,R2,R5,R50,R51,R200,R209,R218,R227,R236,R245,R257 (4.7k_R0603-4) have different partnumbers: C66697 and C1980
* Warning! Identical parts R232,R233,R234 and R421,R424,R425,R757,R762,R763,R767,R768,R102,R108,R109,R121,R127,R128,R141,R142,R757_2,R762_2,R763_2,R767_2,R768_2 (10k_R0402) have different partnumbers: C11616 and C25744
```

Qwerty-OFF, 2023-05-18:
`Warning! Identical parts C753,C754,C759,C100,C101,C105,C108,C109,C110,C111,C112,C113,C118,C119,C120,C121,C122,C124,C126,C127,C128,C129,C130,C134,C147,C170,C175,C180,C185,C190,C195,C204,C205,C206,C207,C217,C218,C220,C221,C222,C230,C231,C232,C233,C270,C275,C280,C285,C753_2,C754_2,C759_2 and C411,C413,C420,C470 (100n_C0402) have different partnumbers: C1525 and C307331`

16V and 50V. Do we need different values for the Value field for such situations? Or use 50V everywhere? The price difference is about $0.3 per board.

`Warning! Identical parts VD100,VD101,VD102,VD103 and D18,D19,D23 (BAT54S_SOT-23) have different partnumbers: C47546 and C408389`

Fixed, similar parts, but different PN

`Warning! Identical parts R125,R170,R171,R175,R176,R180,R181,R185,R186,R190,R191,R196,R197 and R1,R2,R5,R50,R51,R200,R209,R218,R227,R236,R245,R257 (4.7k_R0603-4) have different partnumbers: C66697 and C1980`

5% and 1%. Fixed, now 1% everywhere

`Warning! Identical parts R232,R233,R234 and R421,R424,R425,R757,R762,R763,R767,R768,R102,R108,R109,R121,R127,R128,R141,R142,R757_2,R762_2,R763_2,R767_2,R768_2 (10k_R0402) have different partnumbers: C11616 and C25744`

Wrong PN for 10k in pullups.csv. Fixed

andreika-git, 2023-05-19:
> 16V and 50V. Do we need different values for the Value field for such situations? Or use 50V everywhere? The price difference is about $0.3 per board.

I vote for using 50V caps everywhere, because "12V" can be actually 14.8V (or even higher) which is too close to 16V...