rusefi/alphax-4chan #143: RevH Tacho output not working but changing between 5v/12v

ElDominio, 2023-07-06:
As stated, tacho does not work (no wave on oscope) but switching voltage in Fancy Hardware DOES change voltage.

Was a pin changed from revF to revH to drive the tacho?

rusefillc, 2023-07-14:
@ElDominio do you know resistance of that tach? Do you want me to get same real tachometer for tests?

ElDominio, 2023-07-14:
Test 1: tacho output set to 12v and pin A4

[image]

[image]

Test 2: tacho output using Aux10, internal 4chan

[image]

[image]

ElDominio, 2023-07-20:
Are scopes useful? Any update?

closed 2023-07-29 by ElDominio