rusefi/alphax-4chan #145: adjust main connector

rusefillc, 2023-07-10:
the hole pitch is 4.2/4.2/3.33mm, GERBER is 4.0/4.0/3.5mm

@Qwerty-OFF мы можем сделать что-то универсальное? немного овальные отверстия может быть? я что-то могу намерять своими разъёмами полезного?

[image]

Qwerty-OFF, 2023-07-10:
Если ты сможешь измерить с точностью 0.2мм, давай узнаем сколько там должно быть на самом деле

rusefillc, 2023-07-11:
[delphi_12042017_HCCPHPE24BKA90F-1224853.pdf](https://github.com/rusefi/alphax-4chan/files/12010120/delphi_12042017_HCCPHPE24BKA90F-1224853.pdf)

rusefillc, 2023-07-11:
ха, внезапно он существует официальный и новый и в принципе его можно в JLC заказать

Qwerty-OFF, 2023-07-11:
Сделал отверстия увеличенного размера для совпадения со старым вариантом и с тем что в документе.
@rusefillc Если все будет хорошо, заставь меня переложить модифицированный компонент в kicad6-libraries для дальнейшего использования

Qwerty-OFF, 2023-10-16:
TODO использовать в следующей ревизии символ из библиотеки

rusefillc, 2023-10-17:
https://github.com/rusefi/kicad6-libraries/commit/ccd7bead220f3861fc2b8eb6ab7e77dbb559d2a7

closed 2023-12-09 by Qwerty-OFF