rusefi/alphax-4chan #149: Pin D3 Digital in 2, is processor pin correct?

ElDominio, 2023-07-28:
I have a car using D3 Digital In 2 as AC input, i can see 12v on AC request and no voltage on no request, so everything is reaching ECU correctly, but I see no change in software.

rusefillc, 2023-07-28:
[image]

[image]

[image]

rusefillc, 2023-07-28:
^^^ does not match yaml

rusefillc, 2023-07-28:
@ElDominio it's turning ugly :( see https://github.com/rusefi/rusefi/commit/baaec6ff328077994a03cd216287685844d2a026

@Qwerty-OFF FYI

closed 2023-07-28 by rusefillc