rusefi/alphax-4chan #150: Pin E2 Out NOS not working, correct processor pin?

ElDominio, 2023-07-28:
I am using for AC relay control. I can see 12v from the relay when blower is on, but no software change drives the transistor. Is the processor pin correct?

rusefillc, 2023-07-28:
[image]

rusefillc, 2023-07-28:
[image]

rusefillc, 2023-07-28:
https://github.com/rusefi/rusefi/blob/162507ca0d06c26ecb05785335fa6bf32e597dbb/firmware/config/boards/hellen/alphax-4chan/connectors/main.yaml#L296

@Qwerty-OFF please confirm?

Qwerty-OFF, 2023-07-28:
Yes that's right

rusefillc, 2023-07-28:
@ElDominio we do not see any problems here need more info

rusefillc, 2023-08-02:
@ElDominio any details?

closed 2023-08-09 by ElDominio